NeuronRobotics/mribot-issues #15: Motor stutter with simple P controller

madhephaestus, 2015-02-03:
When the p controller is running from a far away position for the setpoint there seems to be a stutter. The printouts from the axis shows a inverted value during the stutter. 

Tested is a loop that sends the p controller from 10000 to -10000, waiting long enough for it to settle, and no shudders are visible or audible.

madhephaestus, 2015-02-03:
changing to +/- 100000

acamilo, 2015-02-03:
setpoints changed to 100000 +-

madhephaestus, 2015-02-04:
How is this coming?

acamilo, 2015-02-05:
Tried to get LVDS working in XPS. Gave up, posted to xilinx forum.

noticed oscilator was 100mhz not 50mhz which means the SPI peripheral on the card was running way to fast (pll boost to 200mhz for more oversampling. was at 400mhz) i fixed this and will test soon.

acamilo, 2015-02-06:
1. osc was wrong. max this can go is 217mhz. its running at 200.
2. waveforms looked ugly, cleaned them up by using 75 ohm termination on both sides
3. SPI peripheral was incorrectly configured.

reliability test passing. testing code now.

acamilo, 2015-02-06:
odd.. DAC got kicked into funny mode which required a card reset to recover from.

acamilo, 2015-02-06:
fixed.

acamilo, 2015-02-06:
fuck me. dac gets into a mode where it won't update.

acamilo, 2015-02-06:
1. we don't need 75 ohm termination. this does nothing. removed it.
2. Modified DAC so it runs at a lower clock rate and only updates once per packet
3. Motor controller has a minimum speed below which it won't move. worse still, if you try to run it that slow it locks up and once you crank speed back up it doesn't move and will only recover if you de-assert cw and ccw pins. i added a minimum veloity check to the setCardVelocity function.

acamilo, 2015-02-06:
also, cleaned up wiring on test hardware.

closed 2015-02-06 by acamilo