NeuronRobotics/mribot-issues #17: Velocity set value lock up

madhephaestus, 2015-02-26:
During what should be normal opperation the controllers velocity set is not being executed by the lowest level of the stack. I have validateded the C and Java stack passing a non zero velocity down to the FPGA communication interface and that interface is not setting a velocity to the motor.

madhephaestus, 2015-02-26:
To see what values are passed, uncomment the 2 xil_printf in processSync() in Namespace_HSMRI.c

To run the control loop test, run the main file in MyTest.java from eclipse. This will run the control loop at 100ms so the prints do not destabalize it all. it will dither at that update rate, which is ok, since it will normally run at 5ms. 

This bug is confirmed as part of the FPGA communicaiton system, and not a bad value being passed through the stack.

acamilo, 2015-03-02:
modified carddrv.c
deadzone around zero set by `deadzone`
minimum velocy set by `minvel`

velocity is no longer "clamped at minimum velocity" causing there to be a plateu between min velocity and zero where a change in velocity does not map to a change in voltage. Instead, there is a dead-band and on both sides of the dead-band increasing velocity increases voltage linearly. if dead-band is zero then +/- 1 = minvel +/-2 = minvel +/-2 ... etc with 0=0.

The 100ma fault is caused by jittering around zero causing direction to change twice in 10ms which according to the DS6030's datasheet could cause unknown behavior. setting deadband to 200 stops this from occuring.

closed 2015-03-02 by acamilo